SpiNNakerManchester/SpiNNMan #23: "IndexError: bytearray index out of range" in intermittent cases

AlexRast, 2015-02-13:
Symptoms are: Board is booted. A simulation has already been run. Then, you try to run another simulation.

Message reads as indicated with:
 File "/home/P06/rasta/folders/prg/beta_package_103/spynnaker/pyNN/spinnaker.py", line 112, in run
    self._setup_interfaces(hostname=self._hostname)
  File "/home/P06/rasta/folders/prg/beta_package_103/spynnaker/pyNN/spynnaker_comms_functions.py", line 83, in _setup_interfaces
    self._txrx.discover_scamp_connections()
  File "/home/P06/rasta/folders/prg/beta_package_103/spinnman/transceiver.py", line 718, in discover_scamp_connections
    self._update_machine()
  File "/home/P06/rasta/folders/prg/beta_package_103/spinnman/transceiver.py", line 615, in _update_machine
    chip_0_0_details = ChipInfo(response.data)
  File "/home/P06/rasta/folders/prg/beta_package_103/spinnman/model/chip_info.py", line 60, in __init__
    system_data, 32)
  File "/home/P06/rasta/folders/prg/beta_package_103/spinnman/model/chip_info.py", line 8, in _get_int_from_bytearray
    return _get_int_from_little_endian_bytearray(array, offset)
  File "/home/P06/rasta/folders/prg/beta_package_103/spinnman/_utils.py", line 74, in _get_int_from_little_endian_bytearray
    (array[offset + 1] << 8) | array[offset])
IndexError: bytearray index out of range

The error is intermittent. Resetting the board always cures, or sometimes just running the simulation again.

alan-stokes, 2015-04-08:
So to wrap this error up, it seems that this comes from a currupted packet being sent back to the host from spinnaker (we think). There was a discussion to add a retry to this if it happened, to try to recover from the error without crashing. 

Im not sure if its been implimented, and so will be moving this issue to spynnaker issue tracker for central management.

alan-stokes, 2015-04-08:
moved to https://github.com/SpiNNakerManchester/sPyNNaker/issues/49

closed 2015-04-08 by alan-stokes