SpiNNakerManchester/SpiNNMan #33: BMP Power-on commands unacknowledged when sent to non-board zero boards.

mossblaser, 2015-07-08:
I've discovered a subtle bug in the BMP firmware which caught out Rig (worked-around in project-rig/rig#171) and, from a glance at the code, appears to also be present in SpiNNMan.

When a power-on command is sent to a BMP on a board other than board 0, (i.e. the "p" field of the SCP command is not 0) the board does not return an acknowledgement. Though this is acknowledged as a bug in the BMP firmware by ST, there are no plans to fix it since in practical systems there are no scenarios where board 0 does not exist and thus no reason not to just send the command to board 0.

In SpiNNMan (and Rig, prior to project-rig/rig#171), the command is sent to the first board listed in the set of boards to power on/off:

https://github.com/SpiNNakerManchester/SpiNNMan/blob/master/spinnman/messages/scp/impl/scp_power_request.py#L34 then https://github.com/SpiNNakerManchester/SpiNNMan/blob/master/spinnman/messages/scp/abstract_messages/abstract_scp_bmp_request.py#L33-L42

This means that so long as board 0 is the first board listed in the set of boards to power on, the power on command will work. If board 0 is not listed first (or is not listed at all, e.g. when you just want to power on a single board in the middle of a rack), the BMP bug will result in no acknowledgement being sent back. Though the command will work (the board will turn on), the SCP command will eventually time-out. (As an oddity, the power off command works regardless...).

Incidentally, I may have misinterpreted but be aware that only a subset of the BMP commands apply to more than one board at once, as such accepting more than one board in the general case is probably a bug.

alan-stokes, 2017-08-03:
@lplana is this important to keep/consider?

Christian-B, 2017-08-03:
Luis would rather have a look himself first.

Christian-B, 2017-08-03:
Discussed with Luis.

SpinnMan and all stuff that depends on it only use board 0
Turing off/on or reset without going through board zero is not supported.

No real use case known for adding that functionality.

closed 2017-08-03 by Christian-B