SpiNNakerManchester/SpiNNMan #106: spinnman still has issues with booting larger machines

alan-stokes, 2018-02-14:
without the time.sleep(5) in transceiver line 660 (update machine method). on a 3 board toriod, it regularly boots with missing Ethernet chips on none 0,0 board. 

we need a better solution than a random size of sleep. but this does indicate that we need a better way of knowing when the machine has settled into a usable position.

pabogdan, 2018-03-15:
Is this behaviour:
``
`2018-03-15 08:21:34 ERROR: Error when calling spinn_front_end_common.interface.interface_functions.MachineGenerator.__call__ with inputs {'downed_chips': set([]), 'reset_machine_on_start_up': False, 'hostname': '192.168.240.1', 'scamp_connection_data': None, 'boot_port_num': None, 'bmp_details': None, 'downed_links': set([]), 'max_sdram_size': None, 'downed_cores': set([]), 'auto_detect_bmp': False, 'board_version': 5}

spinnman.exceptions.SpinnmanIOException: IO Error: Failed to communicate with the machine```

what the issue you're reporting looks like? I can't communicate with my torus at all, but a local 4-node board, local single 48-node board and spalloc all work.

alan-stokes, 2018-03-15:
alas not peter. what i would see is a booted machine, but with chip 0 0 not being considered a Ethernet chip. This is a epic blow when you deal with IO speed up stuff. As losing any of the Ethernets is considered a none starter position and the code does not adapt to it. we shouldn't be seeing a machine in those states.  But by adding the sleep, it finished booting properly, and thus had a working machine to run on.

andrewgait, 2019-07-24:
It doesn't look like there's a time.sleep() function in this specific location any more, though there are plenty of other instances of time.sleep() being called during the boot process (in boot_board(...), ensure_board_is_ready(...), _try_to_find_scamp_and_boot(...), to name just three).  [Incidentally, each of those has @alan-stokes' favourite bugbear of using a number when it should probably be using at least a defined constant value...]

Is this still an issue with a 3-board torus (do we even have a 3-board torus to check this out on any more)?

rowleya, 2019-07-24:
I *think* this has been solved with various SCAMP and SpiNNMan changes.  I am guessing that no-one is having issues with this as we would have heard about them lots!  I think this can be closed, but of course re-open if it is still an issue.

closed 2019-07-24 by rowleya