SpiNNakerManchester/SpiNNMan #115: Failure to load executable on board

pabogdan, 2018-03-15:
This example uses spalloc, but it fails with local boards as well.

```
 2018-03-15 08:17:52 ERROR: Error when calling spinn_front_end_common.interface.interface_functions.LoadExecutableImages.__call__ with inputs {'executable_targets': <spinnman.model.executable_targets.ExecutableTargets object at 0x1D0543C8>, 'app_id': 17, 'transceiver': transceiver object connected to 10.11.225.105 with 3 connections}
Traceback (most recent call last):
  File "C:/Work/spinnaker/PyNN7Examples/examples/structural_plasticity_example.py", line 213, in <module>
    sim.run(simtime)
  File "C:\Work\spinnaker\sPyNNaker7\spynnaker7\pyNN\__init__.py", line 229, in run
    globals_variables.get_simulator().run(run_time)
  File "C:\Work\spinnaker\sPyNNaker\spynnaker\pyNN\abstract_spinnaker_common.py", line 313, in run
    super(AbstractSpiNNakerCommon, self).run(run_time)
  File "C:\Work\spinnaker\SpiNNFrontEndCommon\spinn_front_end_common\interface\abstract_spinnaker_base.py", line 829, in run
    self._run(run_time)
  File "C:\Work\spinnaker\SpiNNFrontEndCommon\spinn_front_end_common\interface\abstract_spinnaker_base.py", line 1008, in _run
    self._do_load(application_graph_changed)
  File "C:\Work\spinnaker\SpiNNFrontEndCommon\spinn_front_end_common\interface\abstract_spinnaker_base.py", line 1848, in _do_load
    optional_algorithms)
  File "C:\Work\spinnaker\SpiNNFrontEndCommon\spinn_front_end_common\interface\abstract_spinnaker_base.py", line 1196, in _run_algorithms
    executor.execute_mapping()
  File "C:\Work\spinnaker\PACMAN\pacman\executor\pacman_algorithm_executor.py", line 623, in execute_mapping
    self._execute_mapping()
  File "C:\Work\spinnaker\PACMAN\pacman\executor\pacman_algorithm_executor.py", line 639, in _execute_mapping
    results = algorithm.call(self._internal_type_mapping)
  File "C:\Work\spinnaker\PACMAN\pacman\executor\algorithm_classes\abstract_python_algorithm.py", line 47, in call
    results = self.call_python(method_inputs)
  File "C:\Work\spinnaker\PACMAN\pacman\executor\algorithm_classes\python_class_algorithm.py", line 58, in call_python
    return method(**inputs)
  File "C:\Work\spinnaker\SpiNNFrontEndCommon\spinn_front_end_common\interface\interface_functions\load_executable_images.py", line 27, in __call__
    executable_targets, binary, transceiver, app_id))
  File "C:\Work\spinnaker\SpiNNFrontEndCommon\spinn_front_end_common\interface\interface_functions\load_executable_images.py", line 36, in _launch_binary
    core_subset, binary, app_id, wait=True, is_filename=True)
  File "C:\Work\spinnaker\SpiNNMan\spinnman\transceiver.py", line 1463, in execute_flood
    0x67800000, executable, n_bytes, is_filename=is_filename)
  File "C:\Work\spinnaker\SpiNNMan\spinnman\transceiver.py", line 1885, in write_memory_flood
    nearest_neighbour_id, base_address, reader, n_bytes)
  File "C:\Work\spinnaker\SpiNNMan\spinnman\processes\write_memory_flood_process.py", line 71, in write_memory_from_reader
    self.check_for_error()
  File "C:\Work\spinnaker\SpiNNMan\spinnman\processes\abstract_process.py", line 47, in check_for_error
    raise ex
spinnman.exceptions.SpinnmanGenericProcessException: 
     Received exception class: SpinnmanTimeoutException 
     With message: Operation SCPCommand.CMD_FFD timed out after 1.0 seconds 
     When sending to 255:255:0
     Stack trace: None
```

alan-stokes, 2018-03-15:
looks odd certainly. crashed monitor maybe? how many binaries have you loaded at that point?

pabogdan, 2018-03-15:
No clue, but Andrew's recommended change was to increase 
```
# This is the default timeout when using SCP
SCP_TIMEOUT = 1.0
```
from 1 to some higher value, and this seems to fix it. I guess it could be nice to be able to set this from a config file or something.

It would probably be just as interesting to find out why this needs to be increased in the first place though.

alan-stokes, 2018-03-15:
*shrugs* be one with the SpiNNaker machine. Your welcome to try to debug why that timeout needs to be increased.

Christian-B, 2019-02-08:
Is this still an issue?  Please provide more details or we will close.

pabogdan, 2019-02-08:
I've not experience this issue in a while @Christian-B so something must have fixed it.

closed 2019-02-08 by pabogdan