SpiNNakerManchester/SpiNNMan #136: Execution of binary is unreliable

rowleya, 2018-08-02:
The existing code for executing a binary via SCAMP is unreliable - it is possible that the nearest-neighbour packets used for flood fill are dropped at points in the process.  The missing parts will contain the last binary loaded - on the first run after an app_stop this shows up as a binary "sark" running but with a new app id.  It is worth noting that this happens very rarely.

A possible solution (workaround?) to this would be to retry the flood fill until the binary is no longer "sark" (assuming that the name of the new application is not also sark, which we could outlaw I guess).  The difficulty here is that there is no way currently to count the number of binaries with the given name, so the code would have to read the CPU state on every core (i.e. do the equivalent of ps on all cores).

Another possible solution (maybe also a workaround) is to set the flood fill "count" value to something higher than is currently set (it is currently 0, which means the NN packets are only sent once).

rowleya, 2021-03-25:
This is arguably "fixed" by #163, but that doesn't actually check.  Large simulations seem quite stable now, so this probably doesn't need fixing before 6.0.0, but it might be worth looking at still in the future as simulations get even larger.

dkfellows, 2023-09-26:
Arguably finally resolved by #333

closed 2023-09-26 by dkfellows