SpiNNakerManchester/SpiNNMan #146: MAxCoreID in SpiNNMan

alan-stokes, 2018-12-03:
multiple little things here. 
1. MAx vs Max
2. this is never set by FEC, so is None when goes into SPinnMan via general machine. Nor does spalloc return it, therefore the check in SPiNNMan using this becomes redundant. 

in SpinnMan its _max_core_id which is only used in GetMachineProcess, which uses it as follows:

        if self._max_core_id is not None and max_core_id > self._max_core_id:
            max_core_id = self._max_core_id

so when set to None (which is its only value, its made redundant)

This has ramifications on JavaSpiNNaker (which is what set me off this road int he first place).

Christian-B, 2018-12-03:
I would agree as this gives a false sense of security!

rowleya, 2018-12-03:
I did fix some of this in the "Jenkins" branch (work in progress) here:
https://github.com/SpiNNakerManchester/SpiNNFrontEndCommon/commit/8f2503b1fcef18984fb36fcc475d20d1c5f8cbcd

Not sure if that will fix it all though...

alan-stokes, 2018-12-03:
work in progress is vapour ware my friend. till its mergable. So does your vapour ware remove these? or bring them back into the foreground of usfulness?

rowleya, 2018-12-03:
It will become in progress - the above was a note of "I'm working on it, so you probably don't need to".

alan-stokes, 2018-12-04:
ya doing the java as well?

rowleya, 2018-12-04:
Nope.  This is an issue on SpiNNMan...

alan-stokes, 2018-12-04:
its also in the java. which is what set this off. so one of us in the java support needs to fix it there as well.

dkfellows, 2023-09-26:
The Java code does validation in a bunch of places.

Christian-B, 2024-03-05:
no reference to max_core_id or maxcoreid (case insensitive) found

closed 2024-03-05 by Christian-B