SpiNNakerManchester/SpiNNMan #155: trun of power on spalloc jopbs using spalloc

Christian-B, 2019-03-14:
https://github.com/SpiNNakerManchester/SpiNNMan/blob/master/spinnman/transceiver.py

method _power(...

always uses the bmp_connection to turn or on the power.

But as @lplana points out this could be an issue for several reasons.

1. When you ask for a spalloc job you "own" the boards but not the bmp

2. It only turns of the root board not every board in the job.

We wonder if this is what is causing the 
JobDestroyedError: Machine configuration failed. Error: Requests failed on BMP 10.11.213.0

see for example in.

http://apollo.cs.man.ac.uk:8080/blue/organizations/jenkins/sPyNNaker8%20Integration%20Tests/detail/jenkins2/72/pipeline

Christian-B, 2019-03-14:
At the request of @lplana I am disabling the power save test until this has been addressed.

https://github.com/SpiNNakerManchester/sPyNNaker8/tree/jenkins2/p8_integration_tests/quick_test/test_energy_params_validity/test_power_saving_turned_on_after_discovery

dkfellows, 2019-03-14:
The problem here is not that the Transceiver can manipulate the board's power — this is how the spalloc server works — but rather that it should be the responsibility of the job (and hence a spalloc client request) to do the power control instead of accessing the board directly; the BMP should be considered to be entirely off-limits to any allocation created via spalloc.

Christian-B, 2019-03-14:
So this needs to be _turn_off_on_board_to_save_power and _turn_off_board_to_save_power in AbstractSpinnakerBase

rowleya, 2019-03-14:
This could be caused by 2 things:
 1. The transceiver is attempting to access a BMP that it has not been given i.e. it is determining the BMP IP address from the SpiNNaker IP address.  This should not be allowed, or at least should be possible to block.  If this is happening, it is a bug in the Tranceiver in SpiNNMan.
 1. The Spalloc support in FEC is giving out the BMP details to the transciever.   This should not happen; if it is, it is a bug in FEC.

Christian-B, 2019-03-14:
The first Jenkins run without the power on off test passed with no JobDestoryedError.

Could be a coincidence but....

rowleya, 2019-03-14:
There is an issue, I just don't know where it is.  

The question is to whether it lies in SpiNNMan, FEC or both.  SpiNNMan Transceiver should not allow power off/on if it hasn't been given a BMP to use, and FEC shouldn't give the SpiNNMan Transceiver a BMP to use.  It is possible that one or other of these is happening, which would be one or two bugs!

Christian-B, 2019-03-14:
Found it:
https://github.com/SpiNNakerManchester/SpiNNFrontEndCommon/blob/master/spinn_front_end_common/interface/abstract_spinnaker_base.py
line 2581 is blank when it should be else:

like line 2608

will do a PR.

Christian-B, 2019-03-14:
We have had a full Jenkins run now with the test back on.

Assuming (while crossing fingers) fixed

closed 2019-03-14 by Christian-B

Christian-B, 2019-03-15:
Actually on father testing this was not an issue as the transceiver iterated over the list of bmp connections which was empty so actually did NOTHING!