SpiNNakerManchester/SpiNNMan #158: wait_for_cores_to_be_in_state should have better error on timeout

rowleya, 2019-04-10:
The function wait_for_cores_to_be_in_state sometimes times out.  At this point, it would be good to know which cores it believed were not in the given state after the timeout.  This is currently left to the caller; in some executions by the time the caller lists the cores, they have reached the required state, so it is hard to see what went wrong.  This could be handled by adding a list of the error cores to the exception, so this can be extracted later.

alan-stokes, 2019-05-17:
i have something akin to this in the system executor in my c compressor branch. might be worth seeing if we can generalise it when it gets merged in

Christian-B, 2024-03-05:
Appears already fixed

closed 2024-03-05 by Christian-B