SpiNNakerManchester/SpiNNMan #277: app_id vs DataView

Christian-B, 2022-04-29:
https://github.com/SpiNNakerManchester/SpiNNMan/pull/270 Global_data adds a get_app_id(0 method to get the standard app_id

However the Transceiver methods still require an app_id.

Preferred for when methods normally get data from the view is to make the call with a default None where None means read it from the View.

The issue here is Transceiver.get_multicast_routes which If it gets a None app_id means get routes for all ids.
The only know uses case of this is in quick_test.py.

So few questions here.
1. Should we change transceiver methods to not require an app_id and instead use None to say use SpiNNManDataView.get_app_id()

2. If we make such a change do we do it as part of https://github.com/SpiNNakerManchester/SpiNNMan/pull/270 or in a followup cleanup

3. If we do  it do we still want the ability to get_all_multicast_routes

rowleya, 2022-05-03:
So far as I know, no-one is using the tranceiver for anything much outside of front-end-commmon.  So it is unlikely that anything outside of the view is going to be required.  If we want to allow things where it ignores the app_id, that could alternatively be done with a flag e.g. ```get_all_multicast_routes(app_id=None, ignore_app_id=False)```.  This could of course be left until we have a case that needs it.

Christian-B, 2023-07-10:
the are places where a different app id is used which is why it needs to be passed in.

closed 2023-07-10 by Christian-B